gussmith23/churchroad #13: Wires that are not unioned to anything and then deleted will break the egraph

gussmith23, 2024-03-01:
If the design has a wire that is never assigned to anything, then it will produce an eclass with just a Wire in it. Later, that Wire will be deleted, and anything that pointed to that eclass will now be syntactically incorrect.

gussmith23, 2024-03-01:
I think the issue is that we're generating `(Wire ...)`s for module outputs, when I'm not sure they should have Wires declared for them.

thiskappaisgrey, 2024-03-03:
```
; wire declarations
; $flatten\alu.$abc$171$auto$blifparse.cc:386:parse_blif$172.$not$mycells.v:25$198_Y
(let v0 (Wire "v0" 1))
; $flatten\alu.$abc$171$auto$blifparse.cc:386:parse_blif$172.$or$mycells.v:25$197_Y
(let v1 (Wire "v1" 1))
; $flatten\alu.$abc$171$auto$blifparse.cc:386:parse_blif$173.$xor$mycells.v:9$193_Y
(let v2 (Wire "v2" 1))
; $flatten\alu.$abc$171$auto$blifparse.cc:386:parse_blif$174.$not$mycells.v:25$198_Y
(let v3 (Wire "v3" 1))
; $flatten\alu.$abc$171$auto$blifparse.cc:386:parse_blif$174.$or$mycells.v:25$197_Y
(let v4 (Wire "v4" 1))
; $flatten\alu.$abc$171$auto$blifparse.cc:386:parse_blif$175.$and$mycells.v:13$194_Y
(let v5 (Wire "v5" 1))
; $flatten\alu.$abc$171$auto$blifparse.cc:386:parse_blif$176.$not$mycells.v:3$192_Y
(let v6 (Wire "v6" 1))
; $flatten\alu.$abc$171$auto$blifparse.cc:386:parse_blif$177.$not$mycells.v:25$198_Y
(let v7 (Wire "v7" 1))
; $flatten\alu.$abc$171$auto$blifparse.cc:386:parse_blif$177.$or$mycells.v:25$197_Y
(let v8 (Wire "v8" 1))
; $flatten\alu.$abc$171$auto$blifparse.cc:386:parse_blif$178.$not$mycells.v:25$198_Y
(let v9 (Wire "v9" 1))
; $flatten\alu.$abc$171$auto$blifparse.cc:386:parse_blif$178.$or$mycells.v:25$197_Y
(let v10 (Wire "v10" 1))
; $flatten\alu.$abc$171$auto$blifparse.cc:386:parse_blif$179.$and$mycells.v:13$194_Y
(let v11 (Wire "v11" 1))
; $flatten\alu.$abc$171$auto$blifparse.cc:386:parse_blif$180.$or$mycells.v:25$197_Y
(let v12 (Wire "v12" 1))
; $flatten\alu.$abc$171$auto$blifparse.cc:386:parse_blif$181.$not$mycells.v:25$198_Y
(let v13 (Wire "v13" 1))
; $flatten\alu.$abc$171$auto$blifparse.cc:386:parse_blif$181.$or$mycells.v:25$197_Y
(let v14 (Wire "v14" 1))
; $flatten\alu.$abc$171$auto$blifparse.cc:386:parse_blif$182.$xor$mycells.v:9$193_Y
(let v15 (Wire "v15" 1))
; $flatten\alu.$abc$171$auto$blifparse.cc:386:parse_blif$183.$not$mycells.v:25$198_Y
(let v16 (Wire "v16" 1))
; $flatten\alu.$abc$171$auto$blifparse.cc:386:parse_blif$183.$or$mycells.v:25$197_Y
(let v17 (Wire "v17" 1))
; $flatten\alu.$abc$171$auto$blifparse.cc:386:parse_blif$184.$and$mycells.v:13$194_Y
(let v18 (Wire "v18" 1))
; $flatten\alu.$abc$171$auto$blifparse.cc:386:parse_blif$185.$not$mycells.v:3$192_Y
(let v19 (Wire "v19" 1))
; $flatten\alu.$abc$171$auto$blifparse.cc:386:parse_blif$186.$not$mycells.v:25$198_Y
(let v20 (Wire "v20" 1))
; $flatten\alu.$abc$171$auto$blifparse.cc:386:parse_blif$186.$or$mycells.v:25$197_Y
(let v21 (Wire "v21" 1))
; $flatten\alu.$abc$171$auto$blifparse.cc:386:parse_blif$187.$not$mycells.v:25$198_Y
(let v22 (Wire "v22" 1))
; $flatten\alu.$abc$171$auto$blifparse.cc:386:parse_blif$187.$or$mycells.v:25$197_Y
(let v23 (Wire "v23" 1))
; $flatten\alu.$abc$171$auto$blifparse.cc:386:parse_blif$188.$and$mycells.v:13$194_Y
(let v24 (Wire "v24" 1))
; $flatten\alu.$abc$171$auto$blifparse.cc:386:parse_blif$189.$or$mycells.v:25$197_Y
(let v25 (Wire "v25" 1))
; i_instr
(let v26 (Wire "v26" 6))
; o_out
(let v27 (Wire "v27" 2))

; cells
(union v0 (Op1 (Not) v1))
(let v28 (Op1 (Extract 0 0) v26))
(let v29 (Op1 (Extract 2 2) v26))
(union v1 (Op2 (Or) v28 v29))
(union v2 (Op2 (Xor) v28 v29))
(union v3 (Op1 (Not) v4))
(let v30 (Op1 (Extract 4 4) v26))
(union v4 (Op2 (Or) v30 v2))
(let v31 (Op1 (Extract 1 1) v26))
(union v5 (Op2 (And) v31 v29))
(union v6 (Op1 (Not) v5))
(union v7 (Op1 (Not) v8))
(union v8 (Op2 (Or) v31 v30))
(union v9 (Op1 (Not) v10))
(union v10 (Op2 (Or) v0 v7))
(union v11 (Op2 (And) v6 v9))
(let v32 (Op1 (Extract 0 0) v27))
(union v32 (Op1 (Not) v12))
(union v12 (Op2 (Or) v3 v11))
(union v13 (Op1 (Not) v14))
(let v33 (Op1 (Extract 3 3) v26))
(union v14 (Op2 (Or) v28 v33))
(union v15 (Op2 (Xor) v28 v33))
(union v16 (Op1 (Not) v17))
(let v34 (Op1 (Extract 5 5) v26))
(union v17 (Op2 (Or) v34 v15))
(union v18 (Op2 (And) v31 v33))
(union v19 (Op1 (Not) v18))
(union v20 (Op1 (Not) v21))
(union v21 (Op2 (Or) v31 v34))
(union v22 (Op1 (Not) v23))
(union v23 (Op2 (Or) v13 v20))
(union v24 (Op2 (And) v19 v22))
(let v35 (Op1 (Extract 1 1) v27))
(union v35 (Op1 (Not) v25))
(union v25 (Op2 (Or) v16 v24))

; inputs
(let i_instr (Var "i_instr" 6))
(union v26 i_instr)

; outputs
(let o_out v27)

; delete wire expressions
(delete (Wire "v0" 1))
(delete (Wire "v1" 1))
(delete (Wire "v2" 1))
(delete (Wire "v3" 1))
(delete (Wire "v4" 1))
(delete (Wire "v5" 1))
(delete (Wire "v6" 1))
(delete (Wire "v7" 1))
(delete (Wire "v8" 1))
(delete (Wire "v9" 1))
(delete (Wire "v10" 1))
(delete (Wire "v11" 1))
(delete (Wire "v12" 1))
(delete (Wire "v13" 1))
(delete (Wire "v14" 1))
(delete (Wire "v15" 1))
(delete (Wire "v16" 1))
(delete (Wire "v17" 1))
(delete (Wire "v18" 1))
(delete (Wire "v19" 1))
(delete (Wire "v20" 1))
(delete (Wire "v21" 1))
(delete (Wire "v22" 1))
(delete (Wire "v23" 1))
(delete (Wire "v24" 1))
(delete (Wire "v25" 1))
(delete (Wire "v26" 6))
(delete (Wire "v27" 2))
```

Generated from:
```
module top (
        input logic [5:0] i_instr,
        output logic  [1:0]o_out
        );
    logic [1:0] o_op;
    logic  [1:0]o_a;
    logic  [1:0]o_b;
    decoder decoder(.i_instr (i_instr), 
          .o_op (o_op),
          .o_a (o_a),
          .o_b (o_b)
          );
    alu alu(
           .i_a (o_a), 
          .i_b (o_b),
          .i_control (o_op),
          .o_res (o_out)
          );
endmodule

// The "decoder" is a component that breaks the "input" into smaller stuff
module decoder (
  input logic [5:0] i_instr,
  output logic [1:0] o_op,
  output logic  [1:0]o_a,
  output logic  [1:0]o_b
);
        assign o_op = i_instr[1:0];
        assign o_a = i_instr[3:2];
        assign o_b = i_instr[5:4];
endmodule

module alu (
  input logic [1:0]i_a,
  input logic [1:0]i_b,
  input logic [1:0] i_control, // Control signal to specify the operation
  output logic  [1:0]o_res
  
);
  always_comb  begin
    case (i_control)
      2'b00: o_res = i_a ^ i_b; // Bitwise additon
      2'b01: o_res = ~(i_a | i_b); 
      2'b10: o_res = i_a | i_b; // Bitwise OR
      2'b11: o_res = i_a & i_b; // Bitwise AND
    endcase
  end
endmodule
```

thiskappaisgrey, 2024-03-08:
The issue with this is in the way that the output get's used (in the synth output):
```
  assign o_out[1] = ~ (* src = "top.sv:13.9-18.12|mycells.v:25.12-25.20" *) _25_;
  assign o_out[0] = ~ (* src = "top.sv:13.9-18.12|mycells.v:25.12-25.20" *) _12_;
```

Parts of it gets assigned - but not the entire wire - so the "wire" never gets unioned with anything.

thiskappaisgrey, 2024-03-08:
This roughly translates to this egglog in the IR:
```
(let v32 (Op1 (Extract 0 0) v27)) ; v27 is the output wire
(union v32 (Op1 (Not) v12))
(let v35 (Op1 (Extract 1 1) v27)) ; v27 is the output wire
(union v35 (Op1 (Not) v25))
```
but `v27` (the output wire) itself never gets unioned with anything! Only it's extracts are! We need to propogate the extract up somehow before deletion

thiskappaisgrey, 2024-03-08:
I tired to do this instead:
```
(let o_out v27)
(union v27 (Out "o_out" 2))
```
- but this makes it so that extracting the `out` wire doesn't do anything useful....
I just need an extractable expression for the output wires at the end to get a useful circuit after eqsat.

thiskappaisgrey, 2024-03-08:
Another issue that I want to bring up is that if wires are used in this way (only their extracts are assigned to stuff) - it breaks my module finding because I need to be able to extract the output into 1 expression.

thiskappaisgrey, 2024-03-10:
Proposed solution. 
[image]

Can you help me make a rewrite that does this??!? Or maybe we can just figure out a way to do it in the plugin itself when generating expressions? i.e. when you see 
```
_wire_[x] = ...
```
do something different than what you are doing now.

Here's a picture.

gussmith23, 2024-03-12:
Clarifying in my own words after Thanawat walked me through it:

If we have a wire e.g. an output that is never assigned directly, but is instead assigned "piecewise", i.e. its individual bits are defined, then the output wire will be in an eclass by itself.

Can we automatically determine a non-wire expression to union into that eclass, so that we can then safely delete the wire?

gussmith23, 2024-03-12:
## 2024-03-12

Trying to figure out [#13](https://github.com/uwsampl/churchroad/issues/13).

A key part of the problem is that, if a wire is never *directly* assigned-to, but is only assigned-to "piecewise", then nothing will be unioned with that wire in the egraph, which will cause problems once the wire's deleted and the eclass disappears.

So how do we ensure that something ends up unioned to the wire in the original eclass? Seems like the right thing to do is to explicitly add a `union` per wire created. I.e., in the same way we loop over wires to declare them, we should loop over wires to assign them. That will guarantee we don't skip any wires.

Then, we need to figure out how to correctly assign to each wire such that something ends up in the wire's eclass.

Currently we just iterate through all the wires and declare them at first. Then in a later step, we go through all of the cells, which I think I assumed would end up assigning to all wires. In some sense, it does.

gussmith23, 2024-03-14:
## 2024-03-14

Is there a dependable way to find the wires in the design for which this will be an issue? What characteristics do they have? Alternatively, we could iterate over all wires and redefine them with a concat.

gussmith23, 2024-03-18:
## 2024-03-17

An easy solution is probably to just add one concat expression for every wire.

Minimal example. This module:

```sv
module top(output [1:0] out, input a,b);
  assign out[0] = a;
  assign out[1] = b;
endmodule
```

Results in this Churchroad:

```
; wire declarations
; a
(let v0 (Wire "v0" 1))
; b
(let v1 (Wire "v1" 1))
; { \b \a }
(let v2 (Op2 (Concat) v0 v1))

; cells

; inputs
(let a (Var "a" 1))
(union v0 a)
(let b (Var "b" 1))
(union v1 b)

; outputs
(let out v2)

; delete wire expressions
(delete (Wire "v0" 1))
(delete (Wire "v1" 1))
```

As you can see, the "cells" section is empty. There aren't any cells because there's no gates needed here. However, we're missing the fact that `out` is wired to `a` and `b`.

The question I'm trying to answer right now is: given a wire, how do I see how it's defined? 

Posted here:
<https://github.com/YosysHQ/yosys/discussions/4286>

gussmith23, 2024-03-18:
@thiskappaisgrey can you post the Yosys commands you're running (including the techmapping library file if you're using one) to produce the Churchroad code you posted above?

thiskappaisgrey, 2024-03-18:
```
read_verilog -sv alu.sv;
prep -top alu;
hierarchy -check -top alu;
techmap; opt; dfflibmap -liberty mycells.lib;
abc -liberty mycells.lib;
read_verilog mycells.v;

flatten; opt

write_verilog out/alu_synth.v
write_lakeroad out/alu.egg
```

mycells.lib:
```
library(demo) {
  cell(BUF) {
    area: 6;
    pin(A) { direction: input; }
    pin(Y) { direction: output;
              function: "A"; }
  }
  cell(NOT) {
    area: 3;
    pin(A) { direction: input; }
    pin(Y) { direction: output;
              function: "A'"; }
  }
  cell(AND) {
    area: 4;
    pin(A) { direction: input; }
    pin(B) { direction: input; }
    pin(Y) { direction: output;
             function: "(A*B)"; }
           }
  cell(XOR) {
    area: 4;
    pin(A) { direction: input; }
    pin(B) { direction: input; }
    pin(Y) { direction: output;
             function: "(A'+B')*(A+B)"; }
  }
 
  cell(NOR) {
    area: 4;
    pin(A) { direction: input; }
    pin(B) { direction: input; }
    pin(Y) { direction: output;
             function: "(A+B)'"; }
  }
  cell(DFF) {
    area: 18;
    ff(IQ, IQN) { clocked_on: C;
                  next_state: D; }
    pin(C) { direction: input;
                 clock: true; }
    pin(D) { direction: input; }
    pin(Q) { direction: output;
              function: "IQ"; }
  }
}
```
mycells.v:
```
module NOT(A, Y);
input A;
output Y = ~A;
endmodule

module XOR(A, B, Y);
input A, B;
// output Y = (A & ~B) | (~A & B);
output Y = A ^ B;
endmodule
module AND(A, B, Y);
input A, B;
output Y = A & B;
endmodule

module NAND(A, B, Y);
input A, B;
output Y = ~(A & B);
endmodule

module NOR(A, B, Y);
input A, B;
output Y = ~(A | B);
endmodule

module DFF(C, D, Q);
input C, D;
output reg Q;
always @(posedge C)
	Q <= D;
endmodule
```

gussmith23, 2024-03-18:
Here are my thoughts so far:

- First, I can't reproduce the Churchroad code that Thanawat produced above, because I'm not running the same Yosys passes as him. I'm unsure whether the code I produce has the same errors as his code as I haven't taken the time to dig into it yet.
- Second, in general the issue does seem to be that a Wire is created and never unioned with anything. However, I'm surprised by this behavior. It _seems_ like this can happen when  a wire is defined "piecewise", i.e. all of its bits are assigned-to individually. But in this case, I'm fairly certain the Churchroad backend should never even create a Wire; the code says that, if a signal has "chunks" (i.e. it's defined piecewise), then don't create a named wire for it, but instead express it as the concatenation of its chunks. So it seems like Thanawat's example involves a signal that is defined piecewise but still somehow gets a named Wire associated with it too.

gussmith23, 2024-03-18:
@thiskappaisgrey thanks for the quick reply, will continue pushing on this tomorrow!

gussmith23, 2024-03-19:
# 2024-03-18

Something interesting about Thanawat's example is that it's only v30 that doesn't get unioned to anything. Is it truly the only signal that is defined piecewise? that seems a bit unbelievable to me.

Oh, actually, yeah, it is!
```sv
  assign _00_ = ~ (* src = "mycells.v:3.12-3.14" *) i_control[1];
  assign _01_ = ~ (* src = "mycells.v:25.12-25.20" *) _15_;
  assign _15_ = i_control[0] | (* src = "mycells.v:25.14-25.19" *) i_a[0];
  assign _02_ = i_control[0] ^ (* src = "mycells.v:9.12-9.17" *) i_a[0];
  assign _03_ = ~ (* src = "mycells.v:25.12-25.20" *) _16_;
  assign _16_ = i_b[0] | (* src = "mycells.v:25.14-25.19" *) _02_;
  assign _04_ = ~ (* src = "mycells.v:25.12-25.20" *) _17_;
  assign _17_ = _00_ | (* src = "mycells.v:25.14-25.19" *) i_a[0];
  assign _05_ = _00_ & (* src = "mycells.v:13.12-13.17" *) i_b[0];
  assign _06_ = ~ (* src = "mycells.v:25.12-25.20" *) _18_;
  assign _18_ = _04_ | (* src = "mycells.v:25.14-25.19" *) _05_;
  assign _07_ = ~ (* src = "mycells.v:25.12-25.20" *) _19_;
  assign _19_ = _01_ | (* src = "mycells.v:25.14-25.19" *) _06_;
  assign o_res[0] = ~ (* src = "mycells.v:25.12-25.20" *) _20_;
  assign _20_ = _03_ | (* src = "mycells.v:25.14-25.19" *) _07_;
  assign _08_ = ~ (* src = "mycells.v:25.12-25.20" *) _21_;
  assign _21_ = i_control[0] | (* src = "mycells.v:25.14-25.19" *) i_a[1];
  assign _09_ = i_control[0] ^ (* src = "mycells.v:9.12-9.17" *) i_a[1];
  assign _10_ = ~ (* src = "mycells.v:25.12-25.20" *) _22_;
  assign _22_ = i_b[1] | (* src = "mycells.v:25.14-25.19" *) _09_;
  assign _11_ = ~ (* src = "mycells.v:25.12-25.20" *) _23_;
  assign _23_ = _00_ | (* src = "mycells.v:25.14-25.19" *) i_a[1];
  assign _12_ = _00_ & (* src = "mycells.v:13.12-13.17" *) i_b[1];
  assign _13_ = ~ (* src = "mycells.v:25.12-25.20" *) _24_;
  assign _24_ = _11_ | (* src = "mycells.v:25.14-25.19" *) _12_;
  assign _14_ = ~ (* src = "mycells.v:25.12-25.20" *) _25_;
  assign _25_ = _08_ | (* src = "mycells.v:25.14-25.19" *) _13_;
  assign o_res[1] = ~ (* src = "mycells.v:25.12-25.20" *) _26_;
  assign _26_ = _10_ | (* src = "mycells.v:25.14-25.19" *) _14_;
```

So that's not much help.

Okay, so, here's what I'm confused about: `o_res` doesn't sigmap to a list of chunks. I don't understand why it wouldn't given how it's defined.

How can we figure out how `o_res` is defined?

Okay, here's a summary of what I know so far. Sorry if this is already repeating a lot of what @thiskappaisgrey has already said, but I needed to internalize it for myself.

The basic problem is that, if a wire is assigned to in a piecewise fashion (e.g. `o_res` in the above example which is never assigned to directly `assign o_res = ...` but is instead assigned to in pieces `assign o_res[0] = ...`) then there's never a `(union <the wire> <some other expr>)` command created for the wire, and so when we do `(delete <the wire>)`, we get an empty eclass.

The solution is to more gracefully/correctly handle the case where a signal is defined piecewise. *However,* the real solution actually should cover more cases. Because Verilog allows for a lot more than just `assign o_res[0] = ...`. For example, it also allows (i'm pretty sure) for concatenations on the left hand side: `assign {a, b, c} = ...`. Thus, we need to be handling the more general problem here of `union` commands that can look like `(union <extract expr> <other expr>)` or `(union <concat expr> <other expr>)` or anything else.

Here are three potential solutions to this problem.

1. Handle this at wire creation. This would be an easy way to at least handle the case where a signal is defined piecewise. If, for example, when we were defining the wire for `o_res`, we could see that it is actually defined in two parts `o_res[0]` and `o_res[1]`, then we could create those as separate wires. This is currently what is attempted. However, the method for determining that a signal has multiple parts (or chunks) in Yosys seems unreliable, or I'm doing it wrong. My current strategy is to use a `SigMap` to convert `o_res` to its representation as `{ o_res[0] o_res[1] }`. This works *sometimes,* and I'm not sure why it doesn't always work. I'm wondering if there's some heuristic under the hood of `SigMap` that determines what representation to use. I know it has these heuristics in general, so the question is, is there one for this case. If so, then this approach is going to cause inconsistency. Perhaps there's a better way to know for certain if something is composed of chunks?
2. Handle this at cell creation. That is, we can see when a cell is created whether it's being assigned to an entire wire or just a chunk of a wire. We could do some extra bookkeeping when we see that something is assigned to a chunk, and then later, concatenate all those chunks together. Again, this would work for the piecewise example above, but maybe not for the `assign {a,b,c} = ...` example.
3. Handle this with rewrites/datalog rules after the fact. This is what Thanawat suggested. I like this idea because it may allow us to simplify our program generation further. The problem with this is, how do we actually write the rules? And can we end up with an egraph that still has one thing per eclass in the end? I'm not sure if that's really needed, but it seems like a good idea and I would like to keep it as an invariant for now.

PR #18 (2024-04-01): Fix issue where empty eclasses are generated due to wire deletion
Closes #13