gussmith23/churchroad #31: Strange issue with LUT6

gussmith23, 2024-04-12:
```
`timescale  1 ps / 1 ps
//
// LUT6 primitive for Xilinx FPGAs
// Compatible with Verilator tool (www.veripool.org)
// Copyright (c) 2019-2020 Frédéric REQUIN
// License : BSD
//

module LUT6
#(
    parameter [64:0] INIT = 64'h0000000000000000
)
(
    input  wire I0, I1, I2, I3, I4, I5,
    output wire O
);
    wire [5:0] _w_idx = { I5, I4, I3, I2, I1, I0 };
    
    assign O = INIT[_w_idx];

endmodule
```

produces

```
;;; generated with `yosys -p 'plugin -i churchroad; "read_verilog -sv <lut6.sv> prep -top LUT6; pmuxtree; write_lakeroad"`

; wire declarations
; $auto$splice.cc:140:get_spliced_signal$3
(let v0 (Wire "v0" 2))
; $auto$splice.cc:140:get_spliced_signal$5
(let v1 (Wire "v1" 3))
; $auto$splice.cc:140:get_spliced_signal$7
(let v2 (Wire "v2" 4))
; $auto$splice.cc:140:get_spliced_signal$9
(let v3 (Wire "v3" 5))
; I0
(let v4 (Wire "v4" 1))
; I1
(let v5 (Wire "v5" 1))
; I2
(let v6 (Wire "v6" 1))
; I3
(let v7 (Wire "v7" 1))
; I4
(let v8 (Wire "v8" 1))
; I5
(let v9 (Wire "v9" 1))
; O
(let v10 (Wire "v10" 1))
; _w_idx
(let v11 (Wire "v11" 6))

; cells
(union v11 (Op2 (Concat) v3 v9))
(union v0 (Op2 (Concat) v4 v5))
(union v1 (Op2 (Concat) v0 v6))
(union v2 (Op2 (Concat) v1 v7))
(union v3 (Op2 (Concat) v2 v8))
; 64'0000000000000000000000000000000000000000000000000000000000000000
(let v12 (Op0 (BV 0 64)))
(let v13 (Op1 (Extract 0 0) v12))
(union v10 (Op2 (Shr) v13 v11))

; inputs
(let I0 (Var "I0" 1))
(union v4 I0)
(let I1 (Var "I1" 1))
(union v5 I1)
(let I2 (Var "I2" 1))
(union v6 I2)
(let I3 (Var "I3" 1))
(union v7 I3)
(let I4 (Var "I4" 1))
(union v8 I4)
(let I5 (Var "I5" 1))
(union v9 I5)

; outputs
(let O v10)

; delete wire expressions
(delete (Wire "v0" 2))
(delete (Wire "v1" 3))
(delete (Wire "v2" 4))
(delete (Wire "v3" 5))
(delete (Wire "v4" 1))
(delete (Wire "v5" 1))
(delete (Wire "v6" 1))
(delete (Wire "v7" 1))
(delete (Wire "v8" 1))
(delete (Wire "v9" 1))
(delete (Wire "v10" 1))
(delete (Wire "v11" 6))
```
which seems wrong, because the shift/extract seem to be in the wrong order.

the RTLIL is
```
# Generated by Yosys 0.35+36 (git sha1 c95298225, aarch64-apple-darwin20.2-clang 10.0.0-4ubuntu1 -fPIC -Os)
autoidx 2
attribute \top 1
attribute \src "/Users/andrew/research/lakeroad/verilog/simulation/xilinx-ultrascale-plus/LUT6.v:9.1-21.10"
module \LUT6
  parameter \INIT 64'0000000000000000000000000000000000000000000000000000000000000000
  attribute \src "/Users/andrew/research/lakeroad/verilog/simulation/xilinx-ultrascale-plus/LUT6.v:14.17-14.19"
  wire input 1 \I0
  attribute \src "/Users/andrew/research/lakeroad/verilog/simulation/xilinx-ultrascale-plus/LUT6.v:14.21-14.23"
  wire input 2 \I1
  attribute \src "/Users/andrew/research/lakeroad/verilog/simulation/xilinx-ultrascale-plus/LUT6.v:14.25-14.27"
  wire input 3 \I2
  attribute \src "/Users/andrew/research/lakeroad/verilog/simulation/xilinx-ultrascale-plus/LUT6.v:14.29-14.31"
  wire input 4 \I3
  attribute \src "/Users/andrew/research/lakeroad/verilog/simulation/xilinx-ultrascale-plus/LUT6.v:14.33-14.35"
  wire input 5 \I4
  attribute \src "/Users/andrew/research/lakeroad/verilog/simulation/xilinx-ultrascale-plus/LUT6.v:14.37-14.39"
  wire input 6 \I5
  attribute \src "/Users/andrew/research/lakeroad/verilog/simulation/xilinx-ultrascale-plus/LUT6.v:15.17-15.18"
  wire output 7 \O
  attribute \src "/Users/andrew/research/lakeroad/verilog/simulation/xilinx-ultrascale-plus/LUT6.v:17.16-17.22"
  wire width 6 \_w_idx
  attribute \src "/Users/andrew/research/lakeroad/verilog/simulation/xilinx-ultrascale-plus/LUT6.v:0.0-0.0"
  cell $shiftx $shiftx$/Users/andrew/research/lakeroad/verilog/simulation/xilinx-ultrascale-plus/LUT6.v:0$1
    parameter \A_SIGNED 0
    parameter \A_WIDTH 64
    parameter \B_SIGNED 0
    parameter \B_WIDTH 6
    parameter \Y_WIDTH 1
    connect \A 64'0000000000000000000000000000000000000000000000000000000000000000
    connect \B { \I5 \I4 \I3 \I2 \I1 \I0 }
    connect \Y \O
  end
  connect \_w_idx { \I5 \I4 \I3 \I2 \I1 \I0 }
end
```

ninehusky, 2024-04-23:
Did this get resolved?

gussmith23, 2024-04-23:
Yes, this should have been resolved by #35. Are you finding it to still be a problem?

closed 2024-04-23 by gussmith23