gussmith23/churchroad #76: List of benchmarks from Dan

gussmith23, 2024-06-18:
cc @dpetrisko

I'll use this issue to collect benchmarks from Dan. These should be examples where Vivado produces mappings that Dan suspects we can improve upon.

dpetrisko, 2024-06-18:
# Categories
## BRAMs
* 1rw bit mask (synthesizes to LUT RAM), fixed by 1r1w transformation
* 1rw byte mask (synthesizes to N/8 BRAM), fixed by vivado template
* 2 1RW RAMs -> 1 2R2W BRAM
* 2 1R1W RAMs -> 1 2R2W BRAM
You can see fixes for the first two here: https://github.com/bespoke-silicon-group/basejump_stl/pull/681
The latter two I just think would be awesome
## Retiming
Vivado will only transform these by ~1 stage
* DFF-DFF-DFF-OP
* OP-DFF-DFF-DFF
## Integer Multiplication
Important calculations, should be done in minimal DSPs
* 32x32->32L
* 32x32->32H
* 32x32->64
* 64x64->64L
* 64x64->64H
* 64x64->128
## FP Multiplication
Important calculations, should be done in minimal DSPs
* IEEE SP (1S, 8E, 23F)
* IEEE DP (1S, 11E, 53F)
* Recoded SP (1S, 9E, 23F)
* Recoded DP (1S, 12E, 53F)

gussmith23, 2024-06-19:
which part number are we targeting? in fact, any TCL you could post here (however rough) would help me immensely, bc I'm gonna start replicating these!

dpetrisko, 2024-06-19:
Buckle up :) 

Utility scripts:
https://github.com/black-parrot-hdk/zynq-parrot/tree/master/cosim/tcl

Parts:
https://github.com/black-parrot-hdk/zynq-parrot/blob/0c8b72492bf5c5fcd73db2a9b4c37e058d89905e/cosim/mk/Makefile.env#L24

BlackParrot example:

https://github.com/black-parrot-hdk/zynq-parrot/tree/master/cosim/black-parrot-minimal-example/vivado

dpetrisko, 2024-07-08:
@gussmith23 Added some extra benchmarks (barrel shifting). Unsure if vivado does well or not on these, but it's something that FPGA is much much worse than ASIC by virtue of resource distribution

dpetrisko, 2024-08-26:
Related to the IEEE benchmarks above, these two modules (I think) can be translated to DSP:
https://github.com/bsg-external/HardFloat/blob/master/source/fNToRecFN.v
https://github.com/bsg-external/HardFloat/blob/master/source/recFNToFN.v

https://github.com/bsg-external/HardFloat/blob/master/source/mulRecFN.v

Basically, the normal hardfloat flow is

fNToRecFN -> mulRecFN -> recFNToFN